HelTecAutomation/platform-asrmicro650x #1: Documentation incorrectly lists ASR6501 as microcontroller for newer ASR6502 based products

lnlp, 2020-06-06:
According to product descriptions on Heltec site the microcontroller for these newer '02' series products is ASR6502,

PlatformIO documentation (platform and boards documentation) incorrectly lists ASR6501 as microcontroller for these newer '02' series products.

valeros, 2020-06-07:
Hi @lnlp ! Could you please point out what boards are based on `ASR6502` ? I didn't see any mention of `ASR6502` in the Arduino framework. Thanks!

lnlp, 2020-06-07:
Hi @valeros,
With all respect and no pun intended, but I'm flabbergasted about that question.
For writing (PlatformIO) documentation (and maintaining repositories) for  products one would at least expect familiarity with those products and any existing product documentation (and know how to find it). How otherwise can the results be relied upon? 

[This](https://github.com/HelTecAutomation/platform-asrmicro650x) repository contains the code to support te CubeCell products and ASR650x-Arduino in PlatformIO. This requires knowledge about the CubeCell products, ASR650x-Arduino, platform-asrmicro650x and the required development toolchain. I suggest that you contact Heltec / @Heltec-Aaron-Lee for details about their products.

For public CubeCell product information you may check the [README.md](https://github.com/HelTecAutomation/ASR650x-Arduino/blob/master/README.md#instructions) for ASR650x-Arduino and CubeCell product documentation on [Heltec's web site](https://heltec.org/proudct_center/lora/cubecell/) and [Heltec's documentation website](https://heltec-automation-docs.readthedocs.io/en/latest/cubecell/index.html).

valeros, 2020-06-07:
@lnlp Thanks for your remarks. It's not a major release of the platform so there might be some inaccuracies and since PlatformIO documentation is generated automatically based on the board manifests these inaccuracies may unintentionally slip to the docs. For example, let's take a look at [CubeCell Capsule Solar Sensor product page](https://heltec.org/project/htcc-ac02/). In the overview, it's mentioned as ASR6052-based board, but in the tech specs ASR6501 is listed as the LoRa chip. These mistakes don't affect the build process as the `ASR6501` is [hardcoded](https://github.com/HelTecAutomation/ASR650x-Arduino/blob/master/platform.txt#L21) as the main chip for all boards.

Anyway thanks for pointing this out.

lnlp, 2020-06-07:
Okay clear, so most of the PlatformIO docs is auto generated. It still should be correct though.

What I miss (at least in/I only checked the preprocessor section) are definitions like
-DARDUINO_{build.board} -DARDUINO_ARCH_{build.arch}.
Not that this would make a difference for the issue at hand because the architecture is (or at least should be) 'ASR650x' which currently serves both ASR6501 and ASR6502.

> In the overview, it's mentioned as ASR6052-based board, but in the tech specs ASR6501 is listed as the LoRa chip.
Yes, those and other errors should be corrected as well.

> the ASR6501 is hardcoded as the main chip for all boards

That probably has to do with certain compiler settings that are not effected by how many, which and what type of GPIO ports are actually available on a device.

What's not making things more clear in naming is that the ASR6501 and ASR6502 chips are actually SiP's that combine both a Cypress Cypress PSoC 4000 ARM Cortex microcontroller with Semtech SX1262 LoRa tranceiver. But for reference purposes it is sufficient to mention ASR6501 and ASR6502, because that is what these chips are after all.
On some places where the compiler/preprocessor option '6501' or 'ASR6501' is used this may be the result of poor naming (in which case ASR6500 or ASR650x would have been more appropriate). But such compiler/preprocessor options are (as shown) not representative for the actual ASR650x (ASR6501 or ASR6502) chip being used and should not lead to incorrect documentation. And there even should be some setting somewhere that identifies the real (*01 or *02) chip that is being used because you want to be able to check that later on in (application) code.

valeros, 2020-06-07:
> What I miss (at least in/I only checked the preprocessor section) are definitions like
-DARDUINO_{build.board} -DARDUINO_ARCH_{build.arch}.
Not that this would make a difference for the issue at hand because the architecture is (or at least should be) 'ASR650x' (or 'ASR6500') which currently serves both ASR6501 and ASR6502.

Arduino core for ASR650x is an open source project, if you do see that something is missing in the docs/framework/etc why don't you provide a PR? I'm sure the maintainers will greatly appreciate your contribution.

> In certain places where the compiler/preprocessor symbols '6501' or 'ASR6501' are specified this may be the result of poor naming (in which case ASR6500 or ASR650x would have been more appropriate). But such compiler/preprocessor options are (as shown) not representative for the actual ASR6501 or ASR6502 chip being used and should not lead to incorrect documentation.

> There also should exist an identifier which identifies the actual chip (ASR6501 or ASR6502) for which the code is compiled. Because you want to be able to check that as a dependency in (application) code.

We simply strive to reproduce the same build environment specified in the `platform.txt` file for each platform supported by PlatformIO with no or minimal intrusion from our side that might be caused by differences in the internal build systems. We have no accurate view or sound reasoning behind the decision to hardcode `ASR6501` chip in the `platform.txt` file. This probably will be fixed in the next relase of Arduino core for ASR650x.

lnlp, 2020-06-07:
> if you do see that something is missing in the docs/framework/etc why don't you provide a PR

This was just a quick observation based on the link you provided.

While it would be nice if everyone would fix issues that they encounter and then provide a PR this is not how it works in practice. Putting efforts in incorrectly reporting/documenting the issue already is an important step. I don't feel need to fix every issue that I run into (I could make a 24h job of that 😉), especially in cases where I am user of the product and not developer of the product.

I understand that fixing issues and providing PR's would be greatly appreciated, but taking efforts to report issues should be valued as well as it contributes to insight and (possible) improvement.
(After all, this repository is to support commercial products.)

closed 2020-06-07 by valeros

lnlp, 2020-06-08:
Hi @valeros 
In my first response ("I'm flabbergasted") I was assuming that you are from/related to Heltec (because Heltec repository) hence my surprise about asking me for Heltec product information.
I just noticed that you are from/related to PlatformIO instead which makes things more clear.
If that would have been clear from the start my response would have been different.

lnlp, 2020-06-08:
@valeros @ivankravets 
This issue has been closed while it has not been fixed and should therefore be opened again.

valeros, 2020-06-08:
Doc pages for this platform has been updated https://docs.platformio.org/en/latest/boards/index.html#asr-microelectronics-asr605x

lnlp, 2020-06-08:
I cleared my browser cache but the following pages still list ASR6501 as MCU for all boards:

https://platformio.org/platforms/asrmicro650x
https://platformio.org/boards (enter CubeCell as board name)

Further: search on MCU finds ASR6501 boards only, none ASR6502.

Also searching

valeros, 2020-06-08:
Thanks, the platform registry is now updated as well.

juanjole, 2021-12-22:
Hello, regarding this issue, I think the [Heltec CubeCell Capsule Solar Sensor](https://heltec.org/project/htcc-ac02/) is not well defined and the microcontroller is the ASR6501.

In addition, I am working with this device and I had to hardcode 'asr6501' in the 'mcu' field of my environment to make it work because if not there is a compilation error.

I hope you can fix it in the [code](https://github.com/HelTecAutomation/platform-asrmicro650x/blob/develop/boards/cubecell_capsule_solar_sensor.json) and official documentation of PlatformIO. Thanks.

commit b7a1eb2c (2020-06-07): Update board manifests // Resolve #1

Note that preprocessor symbols are identical for both ASR6051 and ASR6052