HelTecAutomation/platform-asrmicro650x #3: Join after reset

rony-vargas, 2020-10-28:
I am trying to avoid a join every time the device resets and I am using this code line:
`bool keepNet = LORAWAN_NET_RESERVE;`

and `platformio.ini` this flag is set: `board_build.arduino.lorawan.net_reserve = ON`

But it doesn't take any effect, every reset the device goes and re-join

valeros, 2020-10-28:
Hi @rony-vargas ! Does it work in Arduino IDE?

rony-vargas, 2020-10-28:
Hi @valeros  It doesn't work on Arduino IDE as well:
Arduino IDE Version: 1.8.13
When looking to debug prints, device prints `+KeepNet=0`
[image]
But I have confirmed it is ON on both, Arduino and PlatformIO
[image]

platformio.ini
```
[env:cubecell_board_plus]
platform = asrmicro650x
board = cubecell_board_plus
framework = arduino
lib_deps = sabas1080/CayenneLPP@^1.1.0
upload_port = COM17
monitor_speed=115200
board_build.arduino.lorawan.region = US915
board_build.arduino.lorawan.adr = ON
board_build.arduino.lorawan.debug_level = FREQ_AND_DIO
board_build.arduino.lorawan.at_support = ON
board_build.arduino.lorawan.uplinkmode = UNCONFIRMED
board_build.arduino.lorawan.net_reserve = ON
```

valeros, 2020-10-28:
Since the problem is not specific solely to PlatformIO, please forward the issue here https://github.com/HelTecAutomation/ASR650x-Arduino/issues where it should get more attention.

closed 2020-10-28 by valeros