HelTecAutomation/platform-asrmicro650x #6: [Help] Unable to get a working firmware on any Cubecell module with Platformio

thucar, 2021-01-13:
I can build the example firmwares on Platformio just fine, the upload also works but the modules just do not boot up. The only feedback I get over serial is 

> Copyright @ 2019 Heltec Automation.All rights reserved.

This is my platformio.ini for a HTCC-AB02S:

```
[env:cubecell_gps]
platform = asrmicro650x
framework = arduino
board = cubecell_gps
monitor_speed = 115200
board_build.arduino.lorawan.region = EU868
board_build.arduino.lorawan.adr = ON
board_build.arduino.lorawan.debug_level = FREQ_AND_DIO
```
And I'm testing with https://github.com/HelTecAutomation/platform-asrmicro650x/blob/develop/examples/LoRa/LoRaWAN/LoRaWAN/src/LoRaWan.ino

After uploading the RGB LED does not come on and there is no serial communication besides the above mentioned copyright.

If I use the Arduino IDE to upload the same example, then it works correctly.

valeros, 2021-01-15:
Hi @thucar ! What version of ASR650x-Arduino do you use in the Arduino IDE?

thucar, 2021-01-15:
Hi @valeros 
I've tried with both, the default and with the latest Git version:
```
platform = https://github.com/HelTecAutomation/platform-asrmicro650x.git
board = cubecell_gps
framework = arduino
platform_packages = framework-arduinoasrmicro650x@~1.2.0
```
Also have tried with cubecell_gps and cubecell_node (I currently have HTCC-AB02A and HTCC-AB02S on my desk)
The behavior seems to be exactly the same.

I wonder if the fact that I'm running on Linux could be the culprit?

thucar, 2021-01-15:
Problem solved. It was a rookie mistake with setting up a new Project in PlatformIO. Everything works as its supposed to :)

closed 2021-01-15 by thucar