HelTecAutomation/platform-asrmicro650x #7: Bring platformio up to date

jezdd, 2021-05-17:
Can we bring this library up to date with Arduino one please

valeros, 2021-05-18:
Hi @jezdd ! What do you mean? This platform uses the latest stable Arduino framework v1.2.0.

jezdd, 2021-05-18:
There have been over 50 commits since 1.2.0 - can we not have this platformio framework simply picking up the latest commits? or at least as an option?

There are plenty of important fixes released and these cant be pulled down on platformio as its tied to 1.2.0

wtremmel, 2021-05-19:
Like for example GPS_Air530Z.h does not exist here while it does exist in the Arduino framework. And there are more.
So I second jszdd's request.

valeros, 2021-05-19:
We support only stable versions. I'd recommend you to ask the maintainers here https://github.com/HelTecAutomation/CubeCell-Arduino  to publish a new release.

wtremmel, 2021-05-19:
Perhaps someone more familiar with PlatformIO than me can do a fork? I do not want to be forced to use Arduino IDE for development...

Also - both this repository and the https://github.com/HelTecAutomation/CubeCell-Arduino are under the same account - so don't you have a more direct contact to the maintainers of it?

valeros, 2021-05-19:
It may not be obvious, but I'm from the PlatformIO team. Even though the Arduino framework and this repository are under the same account, the people who work on these projects are actually two different and independent teams. We synchronize our projects every time when a new release is published. This time the changes in the latest revision of the framework will require modifications in the PlatformIO build scripts and that's why we cannot just use the upstream version of the framework and need to wait for a stable release.

jezdd, 2021-05-20:
I've made a request for a new release, just have to keep pushing

closed 2021-05-20 by jezdd