HelTecAutomation/platform-asrmicro650x #9: Is there anyway to run a RTOS with Cubecell??

javierfileiv, 2021-07-15:
Hello, I'm trying to run a RTOS example and there is a lot of information missing regarding the HAL of Cubecell AB-01 product.

Is there any HAL to program it baremetal or any arduino RTOS like?

Thank you

miathedev, 2021-09-24:
A little is an understatement. Everything is missing - unfortunately. Personally, I would be satisfied to run an application on it without Arduino. Have you already had success with this?

javierfileiv, 2021-09-24:
I didn't even try... I had believed in this board but it showed just a little tinkering board only usable to play like kids with arduino... Not reliable for real product development

miathedev, 2021-09-24:
:/ Maybe the repo owners can give us a better insight. It would be super nice to be able to use these SoCs at a lower level.