HelTecAutomation/platform-asrmicro650x #10: cubecell_board_pro type is not kown by platformio

hansju-11, 2021-07-26:
Processing cubecell_board_pro (platform: asrmicro650x; framework: arduino; board: cubecell_board_pro)
--------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
Error: Unknown board ID 'cubecell_board_pro'
========================================= [FAILED] Took 1.00 seconds =========================================

I'm running platformio:  Core 5.1.1,   Home 3.3.4

valeros, 2021-07-26:
Hi @hansju-11 ! Support for the `cubecell_board_pro` hasn't been released yet, so you need to use the upstream version of the platform:
```ini
[env:cubecell_board_pro]
platform = https://github.com/HelTecAutomation/platform-asrmicro650x.git
framework = arduino
board = cubecell_board_pro

```

hansju-11, 2021-07-26:
Thx for the fast answer.
I will do "an upstream" build,

perfect it is working.

closed 2021-07-26 by hansju-11