HelTecAutomation/platform-asrmicro650x #15: lorawan.debug_level option NONE doesn't work

SebDominguez, 2022-01-24:
Even with `board_build.arduino.lorawan.debug_level = NONE` into `platformio.ini` the LoRaWan debug output is still ON

https://github.com/HelTecAutomation/platform-asrmicro650x/blob/0f2048ca9bf3e57dc4a74c903ea2c8ad10563e22/builder/frameworks/arduino.py#L165

SebDominguez, 2022-01-26:
Just find out that the LoRa libraries inside the Arduino Env Dev for PlatformIO didn't even use the `#define FREQ_PRINTF` nor `#define DIO_PRINTF(format, ...)` defined in [`debug.h`](https://github.com/HelTecAutomation/CubeCell-Arduino/blob/68b01ab7bf04ab2fb4d41a01287dcf4b42defd97/cores/asr650x/board/inc/debug.h)

There are `Serial.print()` statements all over the place... What's the point of having a debugging header if you throw serials outputs everywhere?

ivankravets, 2022-03-16:
I can't reproduce this issue. Please try `pio run -v` and check the verbose output:

```
_DEVEUI_AUTO=0 -DLoraWan_RGB=1 -DLoRaWAN_DEBUG_LEVEL=0
```

See https://github.com/HelTecAutomation/CubeCell-Arduino/blob/master/boards.txt#L98

closed 2022-03-16 by ivankravets