HelTecAutomation/platform-asrmicro650x #17: New release

jezdd, 2022-02-12:
Can we publish a new and up to date release please

Max-Plastix, 2022-02-13:
That would be great!  Specifically we need the v1.4.0 release from https://github.com/HelTecAutomation/CubeCell-Arduino

1rabbit, 2022-03-10:
Agreed, especially when considering v1.3.0 issues

philltran, 2022-03-14:
A v1.4.0 release would be awesome. Let us know how we can help.

ioeides, 2022-03-16:
@Heltec-Aaron-Lee 
could you please address this?

ivankravets, 2022-03-16:
Resolved in https://github.com/HelTecAutomation/platform-asrmicro650x/commit/d7ba4e91b88224b9c2248a01720b4d123d6d0723

Please upgrade to the latest 1.4.0 version of dev-platform https://registry.platformio.org/platforms/heltecautomation/asrmicro650x/versions

closed 2022-03-16 by ivankravets

ivankravets, 2022-03-16:
Happy coding with PlatformIO! 🚀 

P.S: https://community.platformio.org/t/platformio-is-a-ukrainian-project-please-help-us-stop-the-war/26330

jezdd, 2022-03-16:
Thanks, it just occurred to me that you guys will have other things on your mind - our thoughts are with all those in Ukraine right now

philltran, 2022-03-16:
I feel so insensitive. I did not realize Platform.io is Ukrainian. I pray you and your families are safe.