HelTecAutomation/platform-asrmicro650x #22: current version issues

jezdd, 2022-12-17:
Unsure what is going on with versions here, the latest update is applied to the develop branch? why is this?

I am using platform = https://github.com/HelTecAutomation/platform-asrmicro650x.git#develop but its still not picking up as 1.6 release, and of course its the old 1.4 framework release.

Can we get a full release with the latest framework package 1.5 too, we are lagging behind.

Cheers

Jez

eikaramba, 2022-12-19:
yeah start treating platform io releases as a first-tier citizen and not as a "let's wait until someone complains" sideproject.

jezdd, 2022-12-19:
there is zero reason for delays, we need versioning sorting and the "Upgraded CubeCell core to v1.5.0" please

gasagna, 2023-01-25:
Yes, please update the platform.io registry with the new version.